nathankellenicki/node-poweredup #105: Mqtt interface

anthonyangel, 2020-08-19:
Hi, has anyone written some middleware to present an mqtt interface to control a connected device?

Alternatively is there a dashboard with a rest API?

GianCann, 2020-08-19:
With the first version I haved used the Rest API (written from another guy) with Node-Red and MQTT.
I try to find the repository, but I'm not sure if it work also with the latest implementation of @nathankellenicki library.

anthonyangel, 2020-08-19:
So far I've only found https://github.com/Ozafy/rocrail-mqtt-lego-powered-up

GianCann, 2020-08-19:
What I found was a real http server that interfaced with nathan's library.
I'm trying to recover the info (these are experiments done a long time ago)

GianCann, 2020-08-19:
I find it!
https://github.com/MihaZelnik/PoweredUp-REST-API

GianCann, 2020-08-19:
@anthonyangel What's your project?

anthonyangel, 2020-08-19:
@GianCann I'd like to control a Duplo train from [Home Assistant](https://www.home-assistant.io/) -  there's a couple of things I'm thinking of doing with it:
- Control via Google Home / Alexa
- Having the state of the train (e.g. when it runs over an action brick) trigger other hardware (so I can sync the train lights with a lifx bulb, or the speakers with a Sonos)

GianCann, 2020-08-19:
😄  I also use Home Assistant ;)

anthonyangel, 2020-08-19:
I figured that this is possible, so why not try. I'll work out properly what to do with it once it works!

closed 2021-04-06 by nathankellenicki