nathankellenicki/node-poweredup #107: technicmediumhub accelerometersensor not accurate

lightgoleft, 2020-09-26:
Hi,
 
When stationary, the report of accelerometersensor is something like: 4, -21, 1035, which is far from 0, 0, 0. Is there something wrong of parsing the message? I didn't see anything in LEGO Wireless Protocol describing sensor message. 

Thanks.

nutki, 2020-09-26:
An accelerometer will normally read earth gravity when in rest (unless in free fall with no atmosphere or in outer space). 1000 or 1024 is probably the value corresponding to 1g. The values <50 may be due to sensor misalignment with the gravity vector, poor calibration, noise or combination of those. You can verify this by putting the hub on one of it sides which should change the +/-1000 reading to a different axis.

nutki, 2020-09-26:
In practice accelormeters in mobile/embedded devices are rarely used to measure actual acceleration, instead the typical applications are:
* drop detection (via apparent loss of gravity)
* measuring tilt (strictly it is orientation relative to the gravity vector; since gravity appears as constant 1g is it easy to remove other sources with a lowpass filter)
* shake detection (average value after a highpass filter)

nathankellenicki, 2020-10-02:
I believe what @nutki said here is correct. The value you're seeing is the natural force of gravity "pulling" the hub towards the ground. For this value to read zero, the hub would need to be outwith a gravity field - say, travelling on a spaceship to Mars or falling at a rate equal to the force of gravity.

You can verify this by tilting the hub onto any one of its other sides - the values will flip, and the side facing downwards will start reading max value.

closed 2020-10-02 by nathankellenicki