nathankellenicki/node-poweredup #108: Green value in ColorSensor / RGBLED don't match

cpseager, 2020-10-02:
Spot the difference green in the color enums, from the Spike app source code.

ColorSensor
e.black = "0"
e.violet = "1"
e.blue = "3"
e.azure = "4"
e.green = "5"
e.yellow = "7"
e.red = "9"
e.white = "10"
e.noColor = "-1"
e.changed = "-2"  

RGBLED
0 black
1 pink
2 violet
3 blue
4 azure
5 cyan
6 green
7 yellow
8 orange
9 red
10 white

cpseager, 2020-10-02:
looks like Jorge also noticed this a while back https://github.com/JorgePe/BOOSTreveng/blob/master/ColorDistanceSensor.md

corneliusmunz, 2020-12-14:
I faced the same issue in the legoino library: https://github.com/corneliusmunz/legoino/issues/43 I think thats a bug in the sensor firmware which will not be fixed. In the Lego PoweredUp app you can reproduce that issue with the following program. If you feed in directly the color sensor value to the HUB Led, the HUB LED will display CYAN if you detect a green brick. But if you read out only the color, the app will display green for a green brick. 
[image]

nathankellenicki, 2021-08-30:
I do recall this from a long time ago back when we originally started reverse engineering the protocol. I was hesitant to introduce two seperate enums, but I quite like @corneliusmunz 's solution for his library. I'll follow suit.

nathankellenicki, 2021-08-30:
Implemented in v7.0.0

closed 2021-08-30 by nathankellenicki