nathankellenicki/node-poweredup #109: Question: Why querying temperature when instatiating TechnicMediumHub?

tthiery, 2020-10-11:
Hi Nathan,

when I reviewed your library for my [sharpbrick/powered-up](https://github.com/sharpbrick/powered-up) library I made a note why this library [queries temperature on initialization](https://github.com/nathankellenicki/node-poweredup/blob/7852fa58238ff9716bd1426cc19a1fd399504dbf/src/hubs/technicmediumhub.ts#L42) of the hub.

My lib does not do this and works otherwise fine. But maybe this is a stability factor I do not see or something else. So either my lib misses some quality or this call here is not needed. Can you shed some light into this?

nathankellenicki, 2020-12-23:
Hi @tthiery, this is a very good question!

I did a dig through history, and I believe this is legacy code. Before this library worked the way it does now, there were no separate device objects, and hubs had methods to activate motors and read sensor values.

For sensor unique to each hub type, the library explicitly always subscribed to the values! See example - https://github.com/nathankellenicki/node-poweredup/blob/f9cea9a6bf40d3f51efacaf17fb397158a7e47e5/src/lpf2hub.ts#L44

In short, this code should be removed, which I will do. :)

nathankellenicki, 2020-12-23:
Fixed with https://github.com/nathankellenicki/node-poweredup/commit/e016f14059f7dcb2cea9d69d7348bf2f78691219

closed 2020-12-23 by nathankellenicki