nathankellenicki/node-poweredup #110: Unable to connect consistently on Pi 4

jncraton, 2020-10-14:
I've been working on getting this running on a Pi 4, and I'm having difficulty connecting using the example code. I'm using a PUP Hub with a linear motor and train motor attached. This is the output that I see when I run the example code from the readme using `sudo node ./example.js`:

```
Scanning for Hubs...
Discovered HUB NO.4!   
Discovered HUB NO.4! 
Discovered HUB NO.4!
Discovered HUB NO.4!
```

There is a multiple second delay (~6 seconds) between each hub discovery. During that time, `connect()` is attempted and the hub LED becomes solid white, but then goes back to flashing. After many attempts, the hub will usually connect and be controllable.

The following have already been tried:

- Replacing batteries
- Using a different hub
- Updating to latest hub firmware
- Using a different Pi 4
- Using different peripherals
- Stopping scanning before attempting to connect
- Disabling Wifi on the Pi (using /boot/config.txt method)
- Updating all npm packages
- Updating Pi OS using apt

I've looked around for similar issues, and it looks like this may be an underlying issue with Noble:

https://github.com/abandonware/noble/issues/99

Does anyone have ideas on how to begin to address this?

More details
-----------------

The output of `uname -a` is:

```
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
```

The complete output of `sudo DEBUG=* node example.js` is included [here](https://gist.github.com/jncraton/0a1bedb1bc8f5bc2fc3d59dae7039329). In this case, a successful connection was made after 4 attempts.

dlech, 2020-10-14:
I read somewhere recently that there was issues with the Bluetooth firmware on RPi 3/4 and rolling back the firmware fixed the connectivity issues.

jncraton, 2020-10-14:
Thanks for this. I see a somewhat similar firmware issue here:

https://github.com/raspberrypi/firmware/issues/1447

And here:

https://github.com/Hexxeh/rpi-firmware/issues/238

I attempted a rollback to the same firmware version specified in that thread:

```
rpi-update 8382ece2b30be0beb87cac7f3b36824f194d01e9
```

That did not resolve the issue for me. However, I do think it helped. I'm now able to reliable and repeatedly connect to the hub for 8 seconds, disconnect, and reconnect again using `noble` directly:

```js
const noble = require('@abandonware/noble');

noble.on('stateChange', async (state) => {
  console.log("Starting...")
  if (state === 'poweredOn') {
    console.log("Scanning...")
    await noble.startScanningAsync();
  }
});

noble.on('discover', async (h) => {
  if (h['address'] == {hub address here} ) {
    console.log(h['address'])
    await noble.stopScanningAsync();
    await h.connectAsync();
    console.log("Connected")
    await new Promise(r => setTimeout(r, 8000));
    console.log("Disconnecting")
    await h.disconnectAsync();
    await noble.startScanningAsync();
  }
});
```

jncraton, 2020-10-16:
I was able to get this working by rolling back the Bluetooth firmware specifically. If others are trying to solve this, here is the solution that worked for me:

```
wget http://archive.raspberrypi.org/debian/pool/main/b/bluez-firmware/bluez-firmware_1.2-4+rpt2_all.deb
sudo dpkg -i bluez-firmware_1.2-4+rpt2_all.deb
```

I then rebooted to make sure the new firmware was in use.

closed 2020-10-16 by jncraton