nathankellenicki/node-poweredup #112: web_bluetooth.html example - mobile devices not connecting

ronbenetti, 2020-10-20:
Hello

Thank you for this project, it's been a fun learning experience, and useful too.

I setup apache on a standalone computer and it is serving the "web_bluetooth.html" example. 

Everything works from my desktop (MacOS Catalina)
I can access the page using MacOS and Google Chrome, and can connect to 
- move hub
- Technic hub
- Wedo hub
Connection seems fine, the button colors rotate, and I can disconnect.

Problems start on Mobile
I've tested on Moto G and an ASUS phone. 
Pages load but the hubs rarely connect.
- move hub - occassional connects, but does not display any functionalities and does not respond to changes to tilt
- Technic hub - occassional connection, but does not display any functionalities and does not respond to changes to tilt 
- Wedo hub - works well!

The Move and Technic hubs should have the latest firmware. Both work with the lego android apps.

Have you seen something similar? 
Is there a way to enable debug on web?

Thank you

aileo, 2020-10-21:
I wonder if it can be related to [this comment](https://github.com/nathankellenicki/node-poweredup/issues/28#issuecomment-711040402) of another issue where connect is ok on mac laptop but not android phone.

The exposed solution reminds me of [this comment](https://github.com/nathankellenicki/node-poweredup/pull/86/files#r430090202) on [this PR](https://github.com/nathankellenicki/node-poweredup/pull/86). So _maybe_ I broke something...

ronbenetti, 2020-10-21:
Thanks aileo,

it opens another question, how do you build the web module?

I've been trying with the command:  npx tsc, but it results in a lot of errors. 
tsconfig.json maybe out of date, I've modified it to the point where it wills start to build, but the first errors of many, are related to abandonware-noble.d.ts.

Are you using something like vscode to build it?

aileo, 2020-10-21:
You can run `npm run build:browser` to build the browser sources. It requires some dependencies so make sure to run `npm i` first.

It will generate a new `dist/browser/poweredup.js` which is used by the web_bluetooth.html example.

Noble is not used in the browser version of the build, if you have errors related to noble, then you are probably  trying to build the nodejs version.

ronbenetti, 2020-10-22:
You sir are a scholar and a gentleman!

I can build and I'll investigate further later this week.

Thank you

ronbenetti, 2020-10-28:
I think the easiest fix for older phones is to just install Chrome Beta. I can now connect and see the characteristics of the hubs I have access to and capture/process events.

I do wish connection with the hubs was more consistent, sinceI still get failed connections, often on the first attempt, but it is no longer a roadblock to progress.

closed 2020-10-28 by ronbenetti