nathankellenicki/node-poweredup #113: C API for node-poweredup?

dslate1, 2020-11-30:
Hi,

Looks like a cool package. I'm hoping to be able to control LEGO "powered up" components from a Raspberry Pi 3 Model B+. I'm experienced in C and C++  (and various other languages) but not JavaScript, so I wonder if you know of a C or C++ version of your API.

Thanks,

-- Dave Slate

aileo, 2020-11-30:
Hi Dave,

I have found a young repo written in C: [zherczeg/uc-powered-up](https://github.com/zherczeg/uc-powered-up)
It may be missing some parts, but it is open for contributions.

Not directly related, there are 2 others repos that could help:
- [sharpbrick/powered-up](https://github.com/sharpbrick/powered-up), a .NET implementation written in C#
- [corneliusmunz/legoino](https://github.com/corneliusmunz/legoino), written in C for Arduino

dslate1, 2020-11-30:
Thanks aileo, I'll take a look at those.

tthiery, 2020-12-21:
@dslate1 core issue on raspberry pi is the bluetooth stack. Node, .NET, C++ all run there happily, but the bluetooth stack is a challenge. Especially with the full libraries (like nathankellenicki/node-poweredup or sharpbrick/powered-up) which have a full understanding of (and expectation to) the protocol.

dslate1, 2020-12-21:
@tthiery Thanks for your comment. I'm not surprised that bluetooth has problems. When I first got my Mindstorms NXT,  bluetooth connections to the brick were very unreliable. As I recall, multiple firmware mods were required to make it work reasonably well.

nathankellenicki, 2020-12-23:
@tthiery I've read that this fixes the Bluetooth issue with the RPi 4 - https://www.raspberrypi.org/forums/viewtopic.php?t=292113&p=1766154

I haven't had a chance to test it yet, but from what I've read in various GitHub threads, it's promising. :)

closed 2021-04-06 by nathankellenicki