nathankellenicki/node-poweredup #117: Cannot discover control+ 42099

williamledda, 2020-12-27:
Hi, 
Thanks for this great library that look very promising for what I would like to do :)
I have a Lego Technic 42099. Looking at the compatibility table, should be supported but I'm not able to discover the hub.

Here's my code, taken from the examples

```javascript
onst PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

poweredUP.on("discover", async (hub) => { // Wait to discover a Hub
    console.log(`Discovered ${hub.name}!`);
    await hub.connect(); // Connect to the Hub
    const motorA = await hub.waitForDeviceAtPort("A"); // Make sure a motor is plugged into port A
    const motorB = await hub.waitForDeviceAtPort("B"); // Make sure a motor is plugged into port B
    const motorC = await hub.waitForDeviceAtPort("C"); // Make sure a motor is plugged into port C
    console.log("Connected");
    // Just try to connect for now...
});

poweredUP.scan(); // Start scanning for Hubs
console.log("Scanning for Hubs...");
```
I press the button to activate the HUB and the led start flashing white as normal, but nothing happens. I just see "Scanning for Hubs..." on the console. I'm using MacOs Sierra 10.13.6, node v14.15.3. 

Am I doing something wrong? Any idea?

Thanks :)

williamledda, 2020-12-27:
I tested with my raspberry pi 4 and it works! :)
My guess is that the bluetooth device of my macbook pro (early 2011) is not compatible. Correct?

nathankellenicki, 2020-12-28:
Ah, I'm glad you've found a workaround!

I looked up your model of MacBook Pro (Early 2011), and apparently it only supports Bluetooth 2.1. Bluetooth 4.0 (Bluetooth Low Energy) is required to talk to Powered Up hubs.

Apparently it is possible to upgrade it, if you're so inclined - https://bsmulders.com/2016/11/macbook-2011-upgrades/

williamledda, 2020-12-28:
Thanks you for checking 👍 
I was just curios to test the library with Mac, but finally I will continue the raspberry pi.

closed 2020-12-28 by williamledda