nathankellenicki/node-poweredup #118: Self-compiled library never returns promise

Debenben, 2020-12-27:
Using `const PoweredUP = require("node-poweredup");` the function `motorA.rotateByDegrees(20,10).then(function(value) {console.log("test");});` prints "test" after successful completion as expected. 

If I compile node-poweredup myself, everything works as expected except for the promises like above, which never return. To compile I do the following:

```
git clone https://github.com/nathankellenicki/node-poweredup.git
cd node-poweredup
npm install
npm run build
```

When I replace `const PoweredUP = require("node-poweredup");` with `const PoweredUP = require(".");` it behaves differently and the promise never returns and nothing gets printed.

I was not sure if my problem was in relation to this library so I already posted a question here: https://stackoverflow.com/questions/65470276/javascript-promise-never-resolves/65471208#65471208

nathankellenicki, 2020-12-28:
There's quite a bit that could be going wrong here, it's a bit difficult to know where to start.

> When I replace const PoweredUP = require("node-poweredup"); with const PoweredUP = require("."); it behaves differently and the promise never returns and nothing gets printed.

Where is this file located? Inside the `node-poweredup` directory? Or somewhere else?

You also mention you receive the following in your linked Stackoverflow question:

> error TS2794: Expected 1 arguments, but got 0. Did you forget to include 'void' in your type argument to 'Promise'?

What is your host application? Without knowing what you're trying to write (I think saw you mention Electron?) it's difficult, but assuming it's Typescript, there's a version mismatch there. Typescript 4.1 made the previously required argument to `resolve` explicitly required (instead of optional), so it will throw unless you declare the return type of the promise as void when resolving with no parameters.

You can read about it here: https://devblogs.microsoft.com/typescript/announcing-typescript-4-1/#resolves-parameters-are-no-longer-optional-in-promises

A few days ago I myself made the process of bringing these promises up to speed, so I also suspect you're using a slightly older version of this library. I recommend updating to the latest which explicitly sets the return type of these promises to void (including the one you mention in your Stackoverflow question).

Debenben, 2020-12-28:
Thank you a lot for this fast, detailed answer. 

The application I am trying to write is using Electron and Electron-rebuild. It is my first time ever writing something in JavaScript or Typescript, still work in progress and was not intended for publication. I uploaded it anyway so you can have a look: https://github.com/Debenben/babyele. The version of Typescript there seems to be 4.1.3, I have to test what happens if I change it.

Then there is the scenario I used to test. Here I cloned the node-poweredup repository and put a test.js file directly into node-poweredup folder. But you are right, what I cloned got slightly outdated which I did not notice, sorry. I will test again with most up to date version.

Debenben, 2020-12-28:
I checked the versions now, "npm list | grep poweredup" shows version 6.7.0, I believe this is the version I have installed locally which is working. I went to my clone of node-poweredup repository where the master branch version 6.9.0 was checked out, and did
```
node-poweredup$ rm -r -f node_modules
node-poweredup$ git checkout tags/v6.7.0
node-poweredup$ npm install
node-poweredup$ npm run build:all
node-poweredup$ node test.js
```
then it is working. If I checkout tags/v6.9.0 it is not working. I will try to use the older version for now, please let me know if I can do more tests or help to fix it, I guess with my limited knowledge of JavaScript I cannot help much here.

nathankellenicki, 2020-12-28:
Thanks for checking. Can you explain a little more when you say it's not working? Any error messages or anything? I just tested myself using a fresh clone and the latest version (master/v6.9.0) seems to work ok for me.

One thing to check - can you successfully run the examples, eg. `node examples/leds.js`? Those reference the library in a similar fashion as you (using `..` to import relatively).

Debenben, 2020-12-29:
`node examples/leds.js` is working in both versions.

If I take `examples/sample_usage.js` and replace the while loop with `motorA.rotateByDegrees(20,10).then(function(value) {console.log("test");});` in version 6.7.0 it rotates, stops and then prints "test", in version 6.9.0 it rotates and stops and then does not print anything.

I am afraid I also have to correct my statement that it is working in the old version 6.7.0. The simple test is always working, but in my application it sometimes also does not return (it depends a bit on how you move the sliders in my application, so I don't have an easy way how to reproduce this yet). I did some more research there with the old version 6.7.0. It seems to be handled in src/hubs/lpf2hub.ts with the `private _parsePortAction (message: Buffer)` function where it looks for "0x0a" being returned. With my application I sometimes get "0x0c" and "0x0e" there. I found some documentation about this https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#document-4-Buffering and according to that it would be "Idle" +"Current Command(s) Discarded" and "Idle"+"Current Command(s) Discarded"+"Buffer Empty + Command Completed" respectively but there are some things I don't understand about that documentation yet.

Since you were asking about version 6.9.0 I just tried to output the same there and either I did something wrong or the `private _parsePortAction (message: Buffer)` isn't called there.

Debenben, 2021-01-03:
@nathankellenicki Thank you for merging the fix for the wrong constant and sorry to bother you again with this.

The version I am using now has both patch-1 and path-2 (pull request #121) commits included. While both are substantial improvements I am still having trouble with command sequences like:

```
await motorA.rotateByDegrees(...)
await Promise.all( [ motorA.rotateByDegrees(...), motorB.rotateByDegrees(...), ...
await motorB.rotateByDegrees(...)
```
They get stuck somewhere and jump back to life when you try to execute the next sequence which happens to contain the motor where the feedback was missing. Tests like those I wrote in the pull request were all successful, it just happens occasionally with my application. Low battery power of the hub possibly also contributes.

Before spending more time analyzing this I wanted to ask if you or others are also experiencing this? Maybe already have some analysis or solution? Some insights on the official lego documentation would also help, namely:

* The feedback message format https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#port-output-command-feedback allows for three port IDs. Does someone know if and when they are being used? They do not appear to be implemented in node-poweredup.
* How do the startup and completion bits https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#st-comp work? How do they modify the feedback you get?
* Can you get 0x10 bit = busy/full as feedback?

My idea for "the better solution" I was referring to in #121 would be a "safe mode" like this:
* If you try to send a new action command, it would check that the device has `Number(device._busy) === device._finishedCallbacks.length`. If this is not the case it would send `reject("failed to send command")` The user can catch this and try to send later.
* All action commands register their resolve function to `device._finishedCallbacks`. If a command is send, `device._finishedCallbacks.length` is increased by one. If the command is successfully received, the port feedback message increases `Number(device._busy)` accordingly or resolves or rejects callbacks to make them equal again.
* If you get a feedback message only reporting one command but `device._finishedCallbacks.length === 2` you call `reject("failed to receive feedback")` for the first one.
* Whenever a command is sent, you activate a timeout that would call `reject("failed to receive feedback")` if no feedback message is received after a few milliseconds.

The idea would be to make this behavior optional, e.g. you have to call "useSafeMode(true)" on the device. This way you can choose to not limit your response time and throughput when your bluetooth connection is slow. 

The scenario in this proposal which I don't know how to address better is when the hub answers with 0x01 and starts execution, but after completion or failure the finished feedback message is lost. Here you are still stuck with a promise that does not resolve until you send another command.

Debenben, 2021-01-05:
I made a stupid mistake in my application: My speed calculation was wrong and sometimes it gave the command rotateByDegrees with degrees larger than 0 and speed equal to 0 and guess what: It never finished!

I corrected this in my application and the problems are gone, so the patch-2 branch #121 does work reliable after all.

aileo, 2021-01-05:
> * The feedback message format https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#port-output-command-feedback allows for three port IDs. Does someone know if and when they are being used? They do not appear to be implemented in node-poweredup.

I had a look at the LWP and as your PR #121 already handle a lot about your finding, I thought I could suggest this little code to handle multiple port feedback (*it is not tested*):

```typescript
private _parsePortAction (message: Buffer) {
    for (let offset = 3; offset < message.length; offset += 2) {
        const device = this._getDeviceByPortId(message[offset]);

        if (device) {
            device.finish(message[offset+1]);
        }
    }
}
```

> I made a stupid mistake in my application: My speed calculation was wrong and sometimes it gave the command rotateByDegrees with degrees larger than 0 and speed equal to 0 and guess what: It never finished!

the "guess what" made my day

nathankellenicki, 2021-04-06:
Thank you for the above and the PR! I tested it and it's actually solved something that I long thought was a problem but never got around to investigating, that of chaining really quick commands back to back. Much appreciated!

closed 2021-04-06 by nathankellenicki