nathankellenicki/node-poweredup #120: LEGO Mindstorms Robot Inventor - clarification please!

afeltham, 2020-12-30:
Hi there, firstly AMAZING project! Thanks for sharing this. Really looking forward to making somethings with it. 

Could i just get some clarification though to see if i'm doing something wrong, or if its just not possible? I have the LEGO Mindstorms Robot Inventor ( well, my son does! ) and i want to connect to it but the hub never seems to be connected. I can connect to the Lego Mario that he also have so i generally know my system works. I notice in the documentation you say "The SPIKE Prime Hub does not use Bluetooth Low Energy, so is not supported via this library" but at face value that seems to be different from the Lego Mindstorms kit. However, looking at the devices via pictures etc they do look very similar and so i'm essentially wondering if Lego Mindstorms Robot Inventor === SPIKE Prime? Then again i noticed this commit ( https://github.com/nathankellenicki/node-poweredup/commit/afe4eadf0867059b04cea25bd8eec3cd4ff319bd ) which specifically mentioned the Mindstorms motors so i'm a bit confused really. 

I've connected to Mario, Move Hub, Lego Boost - all fine. Robot Inventor - doesn't connect. 

Any ideas?

nathankellenicki, 2020-12-30:
Hi, and thanks for the kind comments!

Unfortunately, what you assume is true - Mindstorms Robot Inventor and SPIKE Prime are the same, minus a couple of firmware variations. As these hubs use Bluetooth Classic, they can't be connected to via Bluetooth Low Energy.

They do however have the capability to talk Bluetooth Low Energy to other devices, so if you were to write Python on your Mindstorms hub, you could connect to Powered Up hubs (Boost, Mario, etc). But note you have to implement LWP3 protocol support yourself.

I think the confusion around the motors/sensor is because technically all Mindstorms/SPIKE and Powered Up motors sensors use the same connector and speak the LPF2 protocol. What this means is that a Mindstorms/SPIKE motor or sensor can be used on a Powered Up hub. This library supports those use cases.

afeltham, 2020-12-30:
Forgive the perhaps ignorant question... if it doesn't use BLE, does that mean that in theory you could use your library but with a different Bluetooth library? If i had another library that could talk the right bluetooth tech, again, could there be a way of fetching the right commands from your library? ie, are constants and parts of the protocol exposed for example?

Thanks for the amazingly quick reply.

nathankellenicki, 2020-12-30:
You could totally use a different library, but it doesn't speak LWP3, so nothing would be transferable from this library.

As an actual fact, when connected via Bluetooth (or USB) the hub exposes a serial port you can connect to at 115200 baud via a terminal, which drops you right into a Python REPL. From there you could implement whatever communications protocol you want (the SPIKE Prime/Mindstorms apps talk JSON over this serial connection).

You can read more here: https://www.instructables.com/MicroPython-on-SPIKE-Prime/

afeltham, 2020-12-30:
So, are you suggesting sending commands / scripts such as : 

```
import hub,utime
accel = hub.motion.accelerometer()
for Acc in accel:
     hub.display.show(str(Acc))
     utime.sleep(0.8)
```

via the serial port? 
sending JSON may actually be slightly better for the situation I had in mind. Is that documented anywhere?

nathankellenicki, 2021-04-06:
Sorry for the late response, I'm unsure if you're still looking for this. Unfortunately I don't have any documented info on the JSON protocol as I haven't done much research into the SPIKE Prime/Robot Inventor set as it's not in my use case. I hope you found some info!

closed 2021-04-06 by nathankellenicki

afeltham, 2021-04-06:
Thank you for the update, and sorry, i should have closed this a little while ago. The library has given me what i needed and you've been very helpful along the way! Thanks!