nathankellenicki/node-poweredup #123: Can not found device which is plugged in before connect hub

MJarolimek, 2021-01-23:
When I manully plug off and plug in device during hub is connected, then callbacks "attach" and "detach" is correctly called and methods like "getDeviceAtPort" work properly. But when device is plug in before hub is connected, then device is not found by any of the function called on hub object. I tried it on your example with just two extra rows:

setInterval(() => {
    const hubs = poweredUP.getHubs(); // Get an array of all connected hubs
    document.getElementById("color").style.backgroundColor = PoweredUP.Consts.ColorNames[color];
    hubs.forEach(async (hub) => {
        const led = await hub.waitForDeviceByType(PoweredUP.Consts.DeviceType.HUB_LED);
        led.setColor(color); // Set the color       
     **const light = await hub.waitForDeviceByType(PoweredUP.Consts.DeviceType.LIGHT);
        light.setBrightness(color*10);**
    })
    color++;
    if (color > 10) {
        color = 1;
    }
}, 2000);

It seems to me to be a very obvious problem, so it is unlikely that it will not be found. But I didn't find it in the issues. And I do not know, what I am doing wrong.

I use your scripts via https://cdn.jsdelivr.net/npm/node-poweredup@latest/dist/browser/poweredup.js.

nathankellenicki, 2021-04-06:
Hi, unfortunatley this is a known bug with some Bluetooth adapters. See here: https://github.com/nathankellenicki/node-poweredup/issues/63

Some adapters don't properly follow the spec and start attempting to deliver messages before the connection has completed. This results in some messages being lost.

If you are 100% *certain* that your desired device will be attached by the time you want to use it, you can use the undocumented hub method `manuallyAttachDevice` to force the attachment of the device. This can help bypass the problem you're experiencing.

nathankellenicki, 2021-06-04:
Closing due to inactivity.

closed 2021-06-04 by nathankellenicki