nathankellenicki/node-poweredup #125: Cannot call any functions on large angular motor attached to Hub!

deslay1, 2021-03-07:
I am using a Lego Technic Hub for the Volvo 6x6 Truck, where I have a one large motor, one XL motor, and one position motor (large grey angular motor). The first two work fine to connect. The example script takes care of that pretty easily. However I can't seem to control the angular motor at all! No functions seem to work. The errors I keep getting are like this:
`UnhandledPromiseRejectionWarning: TypeError: motor.resetZero is not a function`
or for any function like rotateByDegrees(), gotoAngle(), gotoRealZero(), etc.

Did I miss something or is the motor currently not supported?
Code:
```const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

poweredUP.on("discover", async (hub) => {
  console.log(`Discovered ${hub.name}!`);
  console.log(`Battery level: ${hub.batteryLevel}`);

  /* await hub.on("attach", (port, device) => {
      if (port === "A") {
          hub.setMotorAngle("A", 170, 100)
      }
  }) */

  await hub.connect();
  const motor= await hub.waitForDeviceAtPort("A"); // Device ID/Type: 76
  console.log("Connected");
  await hub.sleep(3000);

  while (true) {
    await hub.sleep(2000);
    motor.resetZero(20)
    // motor.goToAngle(60, 50);
    await hub.sleep(2000);
  }
});

poweredUP.scan(); // Start scanning for Hubs
console.log("Scanning for Hubs...");```

remcodesign, 2021-03-27:
Hi, if you want to 'hack' a little bit, do the following: 
_(overriding in some way would be better or just add 'TECHNIC_LARGE_ANGULAR_MOTOR_GREY' to all the device locations)_

_file_ : **app/lib/node/hubs/lpf2hub.js:221**
_method_ : **_parsePortMessage**

_change_: **_(the top of the function)_**

```
 async _parsePortMessage(message) {
        const portId = message[3];
        const event = message[4];
        let deviceType = event ? message.readUInt16LE(5) : 0; // change 'const' to 'let'

	// BEGIN change
	if (message[5] === 76) {
		message[5] = 49; // TECHNIC_LARGE_ANGULAR_MOTOR
	}
	if (deviceType === 76) {
		deviceType = 49; // TECHNIC_LARGE_ANGULAR_MOTOR
	}
	// END change

	if (event === Consts.AlertPayload.ATTACHED_IO) { ...
```

nathankellenicki, 2021-04-06:
Hi, apologies for late response. This is fixed in the latest version.

closed 2021-04-06 by nathankellenicki