nathankellenicki/node-poweredup #127: Problems with two motors and wedo2

pimpelsang, 2021-04-10:
Hi,

I'm using a browser BLE connection in OSX Chrome and having issues. Maybe you know what can cause this right away.

Controlling one motor at a time works perfectly. But setting both motors speed at the same time works only about a third time - on the majority of cases the central unit blinks yellow light and motors very slightly nudge forward.

My code:
https://gist.github.com/pimpelsang/06061ec6c35da1e77980593461812122

HTML page to test:
https://gistpreview.github.io/?06061ec6c35da1e77980593461812122

Example video how it behaves:

nathankellenicki, 2021-04-12:
Hi @pimpelsang,

I did some quick testing this morning and aren't able to reproduce the issue on my end. Can I ask about the status of the batteries you're using?

My Googling suggests that flashing yellow/orange is something to do with an insufficient amount of power/current: https://fccid.io/NPI19071/User-Manual/HW-manual-2877326.pdf

I have heard reports of the WeDo 2.0 hub being a little underpowered for two motors, given that it works off only 2x 1.5v batteries, vs. most of the other hubs in the range that use 6x 1.5v batteries.

Can you verify if this behaviour is the same when using the WeDo 2.0 app instead of node-poweredup?

pimpelsang, 2021-04-13:
Strange thing is that with this android app two motors work just fine. 
https://m.apkpure.com/lego-wedo-2-joystick-controller/com.appybuilder.thanh01_pmt.Wedo2_Joystick_Controller

B'ackground blog post and example scripts 
https://chroniclesofrexbarker.wordpress.com/2016/12/31/appy-hour-build-your-own-lego-wedo-2-0-car-tilttodrive-app-for-android/

pimpelsang, 2021-04-13:
Ok, checked the behavior with the official wedo app.

When running both motors with max power (10) then it kind of works but indeed in the end blinks yellow.
When running on both with power 8 (default) then it works without issues.

With wedo app and speed 10: https://photos.app.goo.gl/79gPgnSHLjse8pyY8
With speed 8 and 1: https://photos.app.goo.gl/LwCBtDfGrRUhz3A76

What I also noticed that the wedo app motor power scale and this library doesn't seem to match - with wedo app the motors lowest speed 1 it spins slow, but with app anything below setPower(40) doesn't spin at all.

pimpelsang, 2021-04-13:
You were right about batteries! I changed them to fresh ones and now both motors work on max speed!

nathankellenicki, 2021-04-16:
Great, I'm glad to hear it's now working!

> What I also noticed that the wedo app motor power scale and this library doesn't seem to match - with wedo app the motors lowest speed 1 it spins slow, but with app anything below setPower(40) doesn't spin at all.

I am curious - what speed range does the WeDo 2.0 app give you to play with? 0-100? 0-10? Or something else?

pimpelsang, 2021-04-20:
Yes the power scale in app is 0-10, where 0 is full stop and 10 is max speed.

pimpelsang, 2021-04-24:
I'll close this issue as it does work

closed 2021-04-24 by pimpelsang