nathankellenicki/node-poweredup #133: Can't do a graceful shutdown (lingering tasks)

silentptr, 2021-06-01:
Using node-poweredup v6.11.0 and NodeJS v10.24.0 on the Raspberry Pi 4B 8GB node isn't able to exit after disconnecting from the hub. I have some simple test code here:

	const log = require('why-is-node-running');
	const PoweredUP = require('node-poweredup');
	const poweredUP = new PoweredUP.PoweredUP();

	poweredUP.on('discover', async (hub) =>
	{
		poweredUP.stop(); // stop scanning otherwise the program loops
		console.log('Discovered ' + hub.name + '.');
		await hub.connect();
		console.log('Connected to ' + hub.name + ' (UUID is' + hub.uuid + ').');
		const motorA = await hub.waitForDeviceAtPort('A');
		
		if (motorA === undefined)
		{
			console.log('motorA === undefined');
		}
		else
		{
			console.log('Got motor A from ' + hub.name + '.');
			motorA.setPower(100);
			console.log('Set motor A power to 100.');
			await hub.sleep(5000);
		}
		
		await hub.disconnect();
		console.log('Disconnected from hub.');
		log();
	});

	poweredUP.scan();
	console.log('Scanning for hubs...');

After the hub disconnect line Node stays open. This is the output of why-is-node-running:

	There are 4 handle(s) keeping the process running

	# SIGNALWRAP
	/home/pi/Powered Up Test/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:98 - process.on('SIGINT', this.onSigIntBinded);
	/home/pi/Powered Up Test/node_modules/@abandonware/noble/lib/noble.js:61               - this._bindings.init();

	# Timeout
	/home/pi/Powered Up Test/node_modules/node-poweredup/dist/node/hubs/basehub.js:276 - global.setTimeout(resolve, delay);
	/home/pi/Powered Up Test/node_modules/node-poweredup/dist/node/hubs/basehub.js:275 - return new Promise((resolve) => {
	/home/pi/Powered Up Test/main.js:22                                                - await hub.sleep(5000);

	# Timeout
	/home/pi/Powered Up Test/node_modules/@abandonware/noble/lib/hci-socket/hci.js:153 - setTimeout(this.pollIsDevUp.bind(this), 1000);

	# TickObject
	(unknown stack trace)

Obviously the API is still usable but being able to do a graceful shutdown of Node while using Powered UP would be nice.

nathankellenicki, 2021-06-04:
Can you explain a little bit more what you mean by "node isn't able to exit"? Ctrl-C doesn't work? `process.exit(0);` doesn't complete?

Noble (the underlying BLE library node-poweredup) uses doesn't provide any other methods to stop or disconnect or tidy up other than `stopScanning()`, which you're already calling by doing `poweredUp.stop();` (https://github.com/noble/noble/blob/master/lib/noble.js)

nathankellenicki, 2021-06-04:
After a small bit of research, it appears that the top handle is an open issue with Noble here: https://github.com/noble/noble/issues/839

The bottom handle someone attempted to raise a PR with Noble with, but it was never merged: https://github.com/noble/noble/pull/577/files

Unfortunately I wouldn't count on either of those issues being fixed any time soon as the project hasn't been very lively as of late. Although I have to admit they haven't caused me any issues, so I'd be curious to hear what problems you're running into.

silentptr, 2021-06-05:
Fair enough, thanks for looking into it.

closed 2021-06-05 by silentptr