nathankellenicki/node-poweredup #137: DuploTrainBaseColorSensor unreliable color detection

thschenk, 2021-07-26:
I recently bought a Duplo train (10874) an I now try to control is using node-poweredup. Everything works great so far, except for the color detection sensor. It just keeps spitting out random colors.

After playing around a bit I have very reliable color detection, when I apply the following change to the code:
In devices/duplotrainbasecolorsensor.ts, I changed the value of the `COLOR` mode like this:

From:
```js
export enum Mode {
  COLOR = 0x00,
  REFLECTIVITY = 0x02,
  RGB = 0x03
}
```
To:
```js
export enum Mode {
  COLOR = 0x01,
  REFLECTIVITY = 0x02,
  RGB = 0x03
}
```
After this change I get very reliable color readings.

I am also using the corneliusmunz/legoino library. There I observe the same pattern. The standard behavior (using a mode 0) gives unreliable color readings, while using mode 1 gives great color detection.

Do you think this is something that should be added to node-poweredup?
I can implement that  For backwards compatibility it would be possible to introduce a new `COLOR_NEW` mode that maps to 1, while keeping the old mode set to 0.
Or do you have a different view?

Regards,

Thijs

thschenk, 2021-07-27:
Also [this documentation](https://github.com/sharpbrick/docs/blob/master/devices/duplotrainbasecolorsensor.md) confirms that mode 1 should be used for the color detection.
```
Mode 0 - Color, intensity of the color
Mode 1 - ColorTag, discrete list of understand colors
Mode 2 - Reflection
Mode 3 - RGB
```

nathankellenicki, 2021-08-30:
@thschenk Thanks for the report, and you're absolutely right. I've published v7.0.0 which fixes this.

closed 2021-08-30 by nathankellenicki

thschenk, 2021-09-03:
Thanks for the fix. It works great!