nathankellenicki/node-poweredup #141: osx monterey m1 not working

dftpnd, 2021-11-08:
osx monterey m1 not working

After upgrading the os, the hub is not visible. talked to apple support for an hour, they could not help. maybe you know what the problem may be

pbochynski, 2021-12-29:
I am on macOs 12.1 now. It still doesn't discover any HUB. I think it is related to this issue: https://github.com/abandonware/noble/issues/225

dlech, 2021-12-29:
Passing the LWP3 service UUID to `noble.startScanning()` should work around the issue. (I haven't tried it with noble, but it worked for a similar Python library.)

See https://github.com/abandonware/noble/issues/225#issuecomment-994974358

Also, please file an issue in the Apple Feedback assistant and link the noble issue. Hopefully if enough people report it, Apple might do something about it.

alin256, 2022-01-09:
I also have some basic issues on 12.1, but since it is on Intel, I posted it in a separate issue: https://github.com/nathankellenicki/node-poweredup/issues/142

alin256, 2022-01-09:
Seems that this might be related to the OS and not the M1 chip:

https://developer.apple.com/forums/thread/694146?login=true&r_s_legacy=true

alin256, 2022-01-09:
> Passing the LWP3 service UUID to `noble.startScanning()`

Can you elaborate in more detail on how to do that in code?

nathankellenicki, 2022-01-19:
Hi, I've just published a new version (8.0.0) that should make this work on Monterey again. As mentioned, passing the service id's to the startScanning function solves the issue.

Initially I had concerns with using both LWP3 and WeDo 2.0 service id's as it should find one or the other, not both, but it doesn't seem to pose any problems.

pbochynski, 2022-01-27:
I can confirm - the new version works! Thanks a lot!

alin256, 2022-05-27:
I can confirm the following setups:

* Intel MacOS 12.3.1 with node16 
* M1 MacOS 12.3  with node16

the discovery works for 

1. Boost
2. Duplo train

I am upgrading the OS and testing against 12.4 later today. If I forget to report the results, feel free to @ me on this post.

alin256, 2022-05-27:
The combination of 

* M1 MacOS **12.4** with node16

and both hubs works as well.