nathankellenicki/node-poweredup #156: Cannot connect to Hub

DylanRKelly, 2022-05-14:
I'm trying to connect to a Hub (88009) by running the _sample_usage.js_ file, but all I get is the error:

      node:internal/modules/cjs/loader:361``
      throw err;
      ^

      Error: Cannot find module 'c:\Users\block\OneDrive\Documents\node-poweredup-1\dist\node\index-node.js'. Please verify that 
      the package.json has a valid "main" entry
          at tryPackage (node:internal/modules/cjs/loader:353:19)
          at Function.Module._findPath (node:internal/modules/cjs/loader:566:18)
          at Function.Module._resolveFilename (node:internal/modules/cjs/loader:919:27)
          at Function.Module._load (node:internal/modules/cjs/loader:778:27)
          at Module.require (node:internal/modules/cjs/loader:1005:19)
          at require (node:internal/modules/cjs/helpers:102:18)
          at Object.<anonymous> (c:\Users\block\OneDrive\Documents\node-poweredup-1\examples\sample_usage.js:1:19)
          at Module._compile (node:internal/modules/cjs/loader:1105:14)
          at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
          at Module.load (node:internal/modules/cjs/loader:981:32) {
        code: 'MODULE_NOT_FOUND',
        path: 'c:\\Users\\block\\OneDrive\\Documents\\node-poweredup-1\\package.json',
        requestPath: '..'
      
I know Java, but I am unfamiliar with how JavaScript and VS Code operates.
Any guidance as to how to fix this issue would be greatly appreciated.

Thanks.

aileo, 2022-05-17:
Hi,

From what I can read, I assume you cloned this repo, if so, it is not built.

In order to build, you must run :

`npm i` to install its dependencies

then

`npm run build:node` to actually build for nodejs

DylanRKelly, 2022-05-22:
I did in fact clone the repo, but running `npm i` didn't seem to do anything. It only caused a lot of errors. 
`npm run build:node` seemed to fail too.

aileo, 2022-05-23:
> `npm i` didn't seem to do anything. It only caused a lot of errors.

It will not work if not properly installed, can you share the logs so we can have a look ?

DylanRKelly, 2022-05-24:
I hope you mean this log.

[2022-05-24T23_18_47_021Z-debug-0.log](https://github.com/nathankellenicki/node-poweredup/files/8766824/2022-05-24T23_18_47_021Z-debug-0.log)

aileo, 2022-05-31:
As explained [here](https://github.com/nathankellenicki/node-poweredup#nodejs-installation), there are some prerequisites to fulfill.

some more details [here](https://github.com/abandonware/noble#windows)