nathankellenicki/node-poweredup #167: Question: how is this._mode determined?

Symbitic, 2022-08-20:
I'm working on a port of node-poweredup to Deno, but I'm having trouble determining where the value of `this._value` in devices such as `tachomotor.ts` comes from. Could someone please explain to me how that value is determined/received from the hub?

Debenben, 2022-08-21:
Tachomotor and the other devices inherit from device.ts where the mode is set in device.subscribe:

https://github.com/nathankellenicki/node-poweredup/blob/master/src/devices/device.ts#L145

There is actually a problem with this, see this pull request:

https://github.com/nathankellenicki/node-poweredup/pull/145

Symbitic, 2022-08-23:
I get that. I mean how does the hub send the value of `mode`?

Debenben, 2022-08-23:
We request the hub to set the mode in hub.subscribe here:
https://github.com/nathankellenicki/node-poweredup/blob/master/src/hubs/lpf2hub.ts#L77
Currently we simply change device._mode after sending the request in the line linked above.

If the hub changes the mode successfully it sends a response which we currently ignore, i.e. this
https://github.com/nathankellenicki/node-poweredup/pull/145/commits/35f91dd4556392c0ab92e9f5a744dde00816b5b5#diff-7cde924ce4c6cd6142127fe084330b917f139bc06ae19f6e6e9fa29fdcf3c51eR162
With the changes in the pull request only the response would change the mode.

closed 2022-08-28 by Symbitic