nathankellenicki/node-poweredup #176: Error: Cannot find module '@abandonware/bluetooth-hci-socket'

AshleyPinner, 2023-02-08:
Versions:
```
F:\Code\twitch-train-control>npm --version
8.19.3

F:\Code\twitch-train-control>node --version
v18.13.0

Windows 11 Pro, 22621.1194
```

Installation:
```
F:\Code\twitch-train-control>npm install node-poweredup --save

added 9 packages, and audited 10 packages in 11s

found 0 vulnerabilities
```

Attempting to run a very basic script:
```JS
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

poweredUP.on("discover", async (hub) => { // Wait to discover a Hub
    console.log(`Discovered ${hub.name}!`);
    await hub.connect();
    const motorA = await hub.waitForDeviceAtPort("A");
    console.log("Connected");
    console.log(motorA);
    console.log(hub);
});
```
Results in this error:
```
F:\Code\twitch-train-control>node traincontrol.js
node:internal/modules/cjs/loader:1042
  throw err;
  ^

Error: Cannot find module '@abandonware/bluetooth-hci-socket'
Require stack:
- F:\Code\twitch-train-control\node_modules\@abandonware\noble\lib\hci-socket\hci.js
- F:\Code\twitch-train-control\node_modules\@abandonware\noble\lib\hci-socket\bindings.js
- F:\Code\twitch-train-control\node_modules\@abandonware\noble\lib\resolve-bindings.js
- F:\Code\twitch-train-control\node_modules\@abandonware\noble\index.js
- F:\Code\twitch-train-control\node_modules\node-poweredup\dist\node\poweredup-node.js
- F:\Code\twitch-train-control\node_modules\node-poweredup\dist\node\index-node.js
- F:\Code\twitch-train-control\traincontrol.js
    at Module._resolveFilename (node:internal/modules/cjs/loader:1039:15)
    at Module._load (node:internal/modules/cjs/loader:885:27)
    at Module.require (node:internal/modules/cjs/loader:1105:19)
    at require (node:internal/modules/cjs/helpers:103:18)
    at Object.<anonymous> (F:\Code\twitch-train-control\node_modules\@abandonware\noble\lib\hci-socket\hci.js:6:28)
    at Module._compile (node:internal/modules/cjs/loader:1218:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1272:10)
    at Module.load (node:internal/modules/cjs/loader:1081:32)
    at Module._load (node:internal/modules/cjs/loader:922:12)
    at Module.require (node:internal/modules/cjs/loader:1105:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    'F:\\Code\\twitch-train-control\\node_modules\\@abandonware\\noble\\lib\\hci-socket\\hci.js',
    'F:\\Code\\twitch-train-control\\node_modules\\@abandonware\\noble\\lib\\hci-socket\\bindings.js',
    'F:\\Code\\twitch-train-control\\node_modules\\@abandonware\\noble\\lib\\resolve-bindings.js',
    'F:\\Code\\twitch-train-control\\node_modules\\@abandonware\\noble\\index.js',
    'F:\\Code\\twitch-train-control\\node_modules\\node-poweredup\\dist\\node\\poweredup-node.js',
    'F:\\Code\\twitch-train-control\\node_modules\\node-poweredup\\dist\\node\\index-node.js',
    'F:\\Code\\twitch-train-control\\traincontrol.js'
  ]
}

Node.js v18.13.0
```

Alright, so what if I require that package explicitly:
```
F:\Code\twitch-train-control>npm install bluetooth-hci-socket

added 4 packages, and audited 14 packages in 5s

found 0 vulnerabilities
```

Running the code again gives me the exact same error as above, though, so no changes if I try this.

package.json after this:
```
{
  "dependencies": {
    "bluetooth-hci-socket": "^0.5.1",
    "node-poweredup": "^8.0.5"
  }
}
```

I can provide the lockfile if required.

I feel like I'm missing something but I do not have enough experience with JS packages to work that out. I'm happy to run debugging commands if you let me know what other output would be useful.

AshleyPinner, 2023-02-08:
I've technically solved this by moving the bluetooth-hci-socket folder in node_modules into the \@abandonware folder, and this has fixed it, though I feel like that's not a recommended solution..,.

Debenben, 2023-02-10:
@abandonware/bluetooth-hci-socket is pulled by @abandonware/noble, and not by node-poweredup. The package is in optional-dependency section there and  recently added to normal dependencies https://github.com/noble/noble/commit/29166df77354275801b5018e9dd5553d258acef5 and then removed again https://github.com/noble/noble/commit/6c84903a586c3deb442417daaf7618ea2557a551

In my setup (linux, oder npm version) the dependency is installed together with node-poweredup. Not sure if the dependency should indeed be optional there, if it would now need to be explicitly requested by node-poweredup and why the behavior is different in different npm versions.

Can you have a look in package.lock file what version of @abandonware/noble you are getting and what happens if you install @abandonware/noble directly?

Debenben, 2023-02-10:
I think it might be this bug https://github.com/npm/cli/issues/4828