nathankellenicki/node-poweredup #185: Devices not connecting to Duplo Base

pdekeulenaer, 2024-04-28:
Hi,

I am trying to get this library to work with my son's Duplo HUB. It seems to detect and connect to the hub just fine, but I never manage to detect the motor, or any other device.

I've done some debugging and best I can tell the hub simply doesn't have any attached devices.

`const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

poweredUP.on("discover", async (hub) => {
        console.log(`Discovered ${hub.name}!`);
        await hub.connect();
 
       console.log('Connected!')
        console.log(`Battery level: ${hub.batteryLevel}`);
        console.log(`Ports: ${hub.ports}`);
        console.log(hub);
 
       if (hub instanceof PoweredUP.DuploTrainBase) {
                console.log('Its a duplo train!');
                console.log('Waiting for motor');
                let devices = await hub.getDevices();
                let motor = await hub.waitForDeviceByType(PoweredUP.Consts.DeviceType.DUPLO_TRAIN_BASE_MOTOR);
                console.log(motor);
        }

poweredUP.scan();

console.log("Scanning...");`

The motor promise never resolves unfortunately. I found an old related issue that said to stop the scanning but that does not seem to fix my issue neither.

When I inspect the output of console.log(hub) I see that the attachedDevices array is empty. Similarly, when I print the content of devices, it's an empty list.

How do I get them to attach?

Debenben, 2024-04-28:
Unfortunately I do not have a Duplo hub. Can you share the log you get with your script? Activating additional debug logging like

```
export DEBUG=*hub
node myscript.js
```
or in windows powershell
```
$env:DEBUG = '*hub'
node myscript.js
```

might be helpful and also the version of node-poweredup and `hub.firmwareVersion`. Do you have other lego hubs, do those work? There is an undocumented function `hub.manuallyAttachDevice(deviceType: number, portId: number)` which you could try, it probably creates a different error but it could help to find the cause of the issue.

pdekeulenaer, 2024-04-28:
I am running it on a Raspberry Pi 5.

Node version v18.20.2
Node-poweredup 9.0.0
Hub firmwareVersion: 1.6
Hub hardwareVersion: 1.0

I don't have any other HUBS to test with. I did try an online script a few weeks ago that used node-poweredup that managed to control my Duplo Train; though I cannot seem to find it back to troubleshoot.

This is the output of the debug 
[debug.log](https://github.com/nathankellenicki/node-poweredup/files/15143247/debug.log)

Will try to manual attach function now

Debenben, 2024-04-28:
Thank you for sharing the logs. 

Your hub does not send Hub Attached I/O messages. Maybe they get lost somwhere, Duplo Train never did or you have a new firmware version? 

For my TechnicHub it looks like this, note the received messages starting with `0f 00 04...` where `04` is https://lego.github.io/lego-ble-wireless-protocol-docs/#hub-attached-i-o:
```
$ export DEBUG=*hub
$ node examples/leds.js
Looking for Hubs...
  technicmediumhub Discovered Control+ Hub +0ms
  technicmediumhub Connecting to Control+ Hub +1s
  lpf2hub LPF2Hub connecting +0ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 01 02 02> +875ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 01 03 05> +192ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 32 01 17 00 00 00 00 10 00 00 00 10> +101ms
  basehub Attached device type 23 (HUB_LED) on port HUB_LED (50) +0ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 3b 01 15 00 00 00 00 10 00 00 00 10> +4ms
  basehub Attached device type 21 (CURRENT_SENSOR) on port CURRENT_SENSOR (59) +2ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 3c 01 14 00 00 00 00 10 00 00 00 10> +1ms
  basehub Attached device type 20 (VOLTAGE_SENSOR) on port VOLTAGE_SENSOR (60) +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 3d 01 3c 00 00 00 00 10 00 00 00 10> +1ms
  basehub Attached device type 60 (TECHNIC_MEDIUM_HUB_TEMPERATURE_SENSOR) on port undefined (61) +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 60 01 3c 00 01 00 00 00 01 00 00 00> +2ms
  basehub Attached device type 60 (TECHNIC_MEDIUM_HUB_TEMPERATURE_SENSOR) on port undefined (96) +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 61 01 39 00 01 00 00 00 01 00 00 00> +1ms
  basehub Attached device type 57 (TECHNIC_MEDIUM_HUB_ACCELEROMETER) on port ACCELEROMETER (97) +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 62 01 3a 00 01 00 00 00 01 00 00 00> +39ms
  basehub Attached device type 58 (TECHNIC_MEDIUM_HUB_GYRO_SENSOR) on port GYRO_SENSOR (98) +40ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 63 01 3b 00 01 00 00 00 01 00 00 00> +2ms
  basehub Attached device type 59 (TECHNIC_MEDIUM_HUB_TILT_SENSOR) on port TILT_SENSOR (99) +2ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 64 01 36 00 01 00 00 00 01 00 00 00> +1ms
  basehub Attached device type 54 (TECHNIC_MEDIUM_HUB_GEST_SENSOR) on port undefined (100) +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 09 00 01 03 06 00 00 00 11> +1ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 01 04 05> +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 09 00 01 04 06 00 00 00 08> +96ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 01 05 02> +1ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 01 06 02> +98ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 06 00 01 05 06 61> +1ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 01 0d 05> +99ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 06 00 01 06 06 64> +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0b 00 01 0d 06 90 84 2b 69 41 1c> +100ms
  lpf2hub LPF2Hub connected +0ms
  technicmediumhub Connect completed +2s
Connected to Technic Hub!
[...]
```
At the end of your log there is a https://lego.github.io/lego-ble-wireless-protocol-docs/#action-types "Hub Will Switch Off" out of nowhere, followed by a connection attempt, I guess it was shut-down by pressing the button and has nothing to do with this issue.

pdekeulenaer, 2024-04-28:
Yeah the hub will switch off comes from a button press.

After further experimentation I can't get the HUB_LED or the SPEAKER to work, even if I manually attach them. I don't really get any (decipherable) error messages neither.

Any further debugging I could do?

Debenben, 2024-05-13:
@pdekeulenaer Were you able to solve the problem? If I had to take a guess I would say https://github.com/abandonware/noble has an issue with the bluetooth stack of the raspberry pi 5.

Things you could do:
* Try with web-bluetooth instead of noble library like in this example https://github.com/nathankellenicki/node-poweredup/blob/master/examples/web_bluetooth.html
* Try it with a different computer
* Try with the previous node-poweredup version 8.0.5 and noble version 1.9.2-15
* Try with an external usb-bluetooth-dongle (I am assuming you use the integrated bluetooth stack of the raspberry)
* Try with the official lego app
* With a TechnicHub you can switch it off, then hold down the button and connect. The official lego poweredup app would then connect to the hub in bootloader mode and update the firmware. In case this is possible with Duplo hub, try to downgrade the firmware.
* export DEBUG=* and see if you can find errors or the missing I/O messages in noble library logs
* use Wireshark to sniff the bluetooth traffic and try to find them there

Maybe @alin256 or @nathankellenicki  have a duplo hub and can help?

pdekeulenaer, 2024-05-14:
I managed to get it to work in the end, just had to figure out the correct port to send the command to. As I couldn't discover the services, it took a bit of digging into some other libraries but ended up with the right config (port 0 for motor, 1 for sound, 17 for light). With these I got my solution to work.

I'll see if I can try the fork to see if this resolves it more structuraly