nathankellenicki/node-poweredup #189: Abandonware/Noble not working on Linux

MrShinyAndNew, 2024-10-16:
There was a change to the Linux kernel that broke abandonware/bluetooth-hci-socket (see https://github.com/abandonware/node-bluetooth-hci-socket/issues/60).

This means node-poweredup doesn't work anymore on Linux. I tried [switching to the Noble fork from stoprocent](https://github.com/MrShinyAndNew/node-poweredup) (see the last comment in that link above) but that doesn't seem to work properly either - the hubs can be discovered, but they rarely connect properly.  For example, the sample code in the readme.md file just prints the "Discovered" message but not the Connected message; then the hub shuts down and then eventually an error ("Command Disallowed") is eventually printed before the program terminates.

kilombo, 2025-04-03:
Any update about this? 

I'm having the same issue, after a system update on Raspberry Pi OS, the library is not able to connect with the Lego Control+ Hub.

Debenben, 2025-06-03:
> Any update about this?
> 
> I'm having the same issue, after a system update on Raspberry Pi OS, the library is not able to connect with the Lego Control+ Hub. Please, I need help with this.

It is fixed in @abandonware/noble version 1.9.2-26, but not in 1.9.2-24 which is requested in https://github.com/nathankellenicki/node-poweredup/blob/master/package.json#L21

To get the new version with the fix you need to override the version requirement. Or use my fork @debenben/node-poweredup