nathankellenicki/node-poweredup #190: Cannot connect to Move Hub

zzbazza, 2024-11-15:
When trying to use with Move Hub (88006) got error:
`libc++abi: terminating due to uncaught exception of type Napi::Error`

```
const PoweredUP = require('node-poweredup');
const poweredUP = new PoweredUP.PoweredUP();

let hub;

async function connectToHub() {
    return new Promise((resolve, reject) => {
        poweredUP.on("discover", async (discoveredHub) => {
            hub = discoveredHub;
            console.log("Hub connecting");
            // console.log(hub);
            try {
                await hub.connect();
                console.log("Hub connected");
                resolve();
            } catch (error) {
                reject("Connection to hub failed");
            }
        });
        poweredUP.scan();
    });
}
```

I was able to connect and work with small hub 88009

fatlard1993, 2025-02-11:
Unsure if its your issue; but something to try is to make sure the hub is up-to-date. You can use the lego powered up app to update it by opening any set (ex. sw bots) and clicking the BT connect button inside there (It'll auto-update once it connects)

I'm able to connect to my Move Hub that is fully up-to-date and using the latest version of this library built with node 20.3

aileo, 2026-11-14:
The Technic Move Hub (88019 in my case) requires encrypted pairing and is not supported by node-poweredup as far as I know. I don't know if it is related to this issue but I think it is worth mentioning.